BitMaker-hub/NerdMiner_v2 #482: ESP32-S3-Zero Support?

ZooloZombie, 2024-09-08:
Trying to make NerdMiner v2 work with the ESP32-S3-Zero
ESP32-S3FH4R2
Dual Core 240 MHz
Built-in 512KB of SRAM and 384KB ROM, onboard 4MB Flash memory and 2MB PSRAM
Octal PSRAM

Specs here:
https://www.waveshare.com/wiki/ESP32-S3-Zero

Purchased from:
https://www.aliexpress.com/item/1005006963045909.html
NOTE:  The RGB LED is on GPIO21 not GPIO10 as indicated on the listing.  Tested this with Arduino IDE / Arduino-ESP32 so the actual product matches the Waveshare version.

I am not sure what I am missing as I have tried various configurations and it keeps getting stuck in a boot loop using the code from the dev branch.  The board seems to be very similar to the M5-StampS3 (no Screen) , Wemos Lolin S3 Mini (4 MB) and Weact S3 Mini (ESP32S3-H4R2).  I am not trying to get a screen working with it, just using it as a miner and monitoring with a pool.  Note there is no UART on the board other than the debugging one for programming built into the ESP32.  I tried various things in platformio.ini trying to make it work, but keep getting stuck in a boot loop once I program the 4 files into the ESP32.  I am programming bootloader to 0x1000, firmware to 0x10000, partitions to 0x8000 and the boot_ap0.bin to 0xe000.  Anyone have any ideas of something obvious I am missing that needs to be a flag in platformio.ini to make it work?  Am I programming to the wrong locations as it is only a 4MB ESP32?  Do I have to do something special in platformio.ini since it has an Octal PSRAM?

Thanks if anyone has some ideas.

julio-matarranz, 2024-09-08:
Dear,
Your bootloader address is wrong for an esp32s3. It should be 0x0000. 0x1000 is for previous esp32.

julio-matarranz, 2024-09-08:
You can duplicate weact in platformio.ini and change RGB_LED_PIN.

ZooloZombie, 2024-09-12:
Ok, so I program the bootloader.bin into 0x0000 , firmware.bin into 0x10000, partitions.bin into 0x8000 and boot_app0.bin into 0xe000 ?  Question related to this, where do I get the boot_app0.bin after doing the "run"?  I seem to get the other files generated in the .pio/build folder after making that change and going through a "run", but that boot_app0.bin one is missing.  The reason I am asking is I did modify it changing the pin and using the pre-build boot_app0.bin from the archive, but it seems the ESP32 just gets warm/hot and I don't see the SSID NerdminerAP being broadcast from other clients.  Also related to the pin, does it want it in GPIO notation or actual chip pin notation?  The pin for the RGB LED is GPIO 21, but the actual corresponding pin from the datasheet ( https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf page 16 ) for GPIO 21 is 27, so I am not clear which the RGB_LED_PIN= is looking for?  Is it GPIO # (21) or actual pin (27)?

Thanks, and sorry for all the questions, I am just  trying to learn more about this.

julio-matarranz, 2024-09-12:
According to waveshare doc  RGB_LED_PIN should be 21.

If you are programming with vs code you should use "upload" or "upload and monitor" from the menu it will be more convenient. You wouldn't have to mind about addresses and boot file.

You can find this file in a platform sub directory or on GitHub releases because I copied it from that sub directory in GitHub actions. 😉

phuthai, 2024-09-16:
i got same problem, any solution please share..

JohnTaipei, 2024-09-28:
choose ESP32-S3 T-QT
it works

phuthai, 2024-09-29:
thank you for your recommended seem it's work., but i really new on this
thing please advise for solve this error. (Loop boot)
look like i have to put configuration file how i can put?

HashBox21, 2025-02-15:
> choose ESP32-S3 T-QT it works

this works with the Zero and other mini S3's?

HashBox21, 2025-02-15:
Anyone figure this all out or have the flash codes for any other S3 mini's?