BitMaker-hub/NerdMiner_v2 #493: Problem with Ideaspark®ESP321.9in ST7789 Display LCD TFT 170x320. No full screen

Pixelssquad, 2024-09-17:
I'm quite a noob in these things, I ran into this problem adapting the program to my esp32 card. not the entire screen is used.
I only changed the file: platformio.ini. Here you can find my little work.
https://github.com/Pixelssquad/NerdMiner_v2_IdeaSparkESP32-1.9in-ST7789/tree/dev
[image]

Pixelssquad, 2024-09-28:
Solved here: https://github.com/Pixelssquad/NerdMiner_v2_IdeaSpark1.9in-ST7789-Display-LCD-TFT-170x320

pt7777, 2024-11-13:
Hi, would the firmware also work with the ideaspark ESP32 1,14 Zoll ST7789 135 x 240 TFT?

Rob-The-Gob, 2024-11-20:
Hi i would like for the 135 x240 too, i have tried to compile but it fails every time

Rob-The-Gob, 2024-11-22:
I HAVE DONE IT!!!!!!!! 
[image]
[image]
[image]

you can download the Project [HERE](https://mega.nz/file/D89R2DIa#sGoMEbcOxMgTjoS3JDbg1Hbh58Pj3bclfEfyLakjYbY)

Rob-The-Gob, 2024-11-23:
> Hi, would the firmware also work with the ideaspark ESP32 1.14in ST7789 135 x 240 TFT?

https://github.com/Rob-The-Gob/ideaspark-esp32-nerdminer-V2

Pixelssquad, 2024-12-27:
tnx to all!

Rob-The-Gob, 2025-02-17:
Hi, click Code and download as zip all bin files are there

https://github.com/Rob-The-Gob/ideaspark-esp32-nerdminer-V2

Would you be able to share the bin files in the Git? I am trying to edit images and not able to find the correct files. TIA!!!

https://github.com/Rob-The-Gob/ideaspark-esp32-nerdminer-V2

Would you be able to share the bin files in the Git? I am trying to edit images and not able to find the correct files. TIA!!!