BitMaker-hub/NerdMiner_v2 #507: Problems setting up the ESP32 S2 mini

MarvinOstwald, 2024-10-13:
I bought an ESP32 s2 mini and tried to flash it with the https://flasher.bitronics.store/ page. But I always get these errors:

> Starting flash process...
> 
> Error connecting to chip: Error: Timeout
> Fetching: github-assets/nerdminer-prerelease-V1.7.0/ESP32-S2-mini-wemos_bootloader.bin
> Fetched: bootloader.bin
> Fetching: github-assets/nerdminer-prerelease-V1.7.0/ESP32-S2-mini-wemos_partitions.bin
> Fetched: partitions.bin
> Fetching: github-assets/nerdminer-prerelease-V1.7.0/boot_app0.bin
> Fetched: boot_app0.bin
> Fetching: github-assets/nerdminer-prerelease-V1.7.0/ESP32-S2-mini-wemos_firmware.bin
> Fetched: firmware.bin
> Flashing failed: Error: Timeout

I am new to all this and unfortunately don't know what I can do to solve the problem. I hope someone here can help me.

gungriffonbr, 2024-12-31:
i have this problem too

Tay21tay, 2024-12-31:
I had mine on for almost a month and found nothing

gungriffonbr, 2025-01-01:
> I had mine on for almost a month and found nothing
> […](#)

you can mine nano or babydoge using unmineable pool have some tutorials on youtube in portuguese and german https://www.youtube.com/watch?v=5ieIyugDIt4&t=1s

R9anSmart, 2025-01-16:
Same here. Unable to flash or erase. ESPWEBTOOL connects, is able to erase and flash. Still not working with manual flash