BitMaker-hub/NerdMiner_v2 #517: HowTo: flash the prerelease 1.7.0 (working for NerdminerV2)

TylonHH, 2024-10-30:
So, as here some issues with flashing the prerelease I like to show how it worked for me.

I have the common ESP-S3 Nerdminer V2 with Display.

1. Go to https://github.com/BitMaker-hub/NerdMiner_v2/releases/tag/nerdminer-prerelease-V1.7.0
2. get **four** files of your device (eg. NerminerV2)
- boot_app0.bin
- NerdminerV2_bootloader.bin
- NerdminerV2_firmware.bin
- NerdminerV2_partitions.bin
3. Open the https://espressif.github.io/esptool-js/ Flash Tool within chrome browser icognito
4. connect your device via USB
5. hit `connect`, select the correct USB Port
6. add the four downloaded files in this order at this adress

`0x0000` bootloader.bin
`0x8000` partitions.bin
`0x10000` firmware.bin
`0xe000` boot_app0.bin

Then it should look like this:
[image]

7. click `Program`
8. when finish -> **IMPORTANT** press the reset button on the side of the device! Not the button near the display.
9. When this worked for you give feedback

stl1988, 2024-11-19:
For v1.7.0, these files don't exist. For v1.6.4, I got it running using the "Nerminer" file over the Nerdminer flasher.

TylonHH, 2024-11-19:
What files doesn't exit? You checked the files under 1st point?

This manual is explicit for prerelease 1.7.0

GitHubUser-2024, 2024-12-04:
Verified just now. Is working **perfectly**. Just maybe add the need to enter boot loader mode.
And instead of device reset I just power cycle it (as my reset button is inaccesible anyway).

throw541, 2024-12-16:
I followed this with the T display version and it doesn't work

Followed this with your instructions and it booted right up

Herty76, 2025-01-21:
Hello, I have a little problem with this.
According to the instructions, the update works so far, and the display also works.

According to the display 76 kH, according to miner stats at Nerdminers.org 0kH.

I also have this phenomenon in two other pools.

Greetings
Mark

stefi01, 2025-01-24:
didnt work for me, said the first file never existed

Herty76, 2025-01-25:
Did you choose the correct files, baud rate 115200?

stefi01, 2025-01-25:
> Did you choose the correct files, baud rate 115200?

hey thanks for the reply, I finely got it working,esp 32 c6 just never worked, I had 3 of those, would flash ok but that's as far as it would go, yes I used 115200, adjusted settings to match above picture, it said some error that the files did not appear to be correct even though they did match the tutorial, I found a youtube video where this guy was having issues too, so but he had a single bin file that did work and its in his description for download, , I tried 10 esp32 boards that I got from one seller on ebay, this error happened to all 10, got some esp 32 c-6's from Amazon but would not even produce their own wifi network,   so last night I checked using another flash programme, it said that all 10 boards do not appear to be genuine, I tried using another esp32 that I had laying around, the 4 files from github again never worked but this one single file from Youtube did, video link [https://www.youtube.com/watch?v=gGPa_K47aL0](url)  hope this makes sense as it sure didn't whilst typing it lol and again, his bin file is in the video description (its just one file but worked)

MrFisk02, 2025-02-01:
Hello everyone Does anyone have a firmware for Nerdminer V3? I accidentally uploaded the firmware, and only then I saw that it was for V2 (I will be grateful for the help!

littymama, 2025-03-19:
I have nerdminer 3 , thought i had installed update to make it go up to 417kh/s. Not much it says nerdminer 2 on it now with some shitty graphics and I'm not sure how to fix it back to right. I should of left it alone as it was happily jammin away at 377kh/s before this
 Any suggestions would be aporeciated. Also I'm not a coder or anything like that but I'm not a slouch either.

Qliams, 2025-03-19:
Hi TylonHH,

I have received mine device just an hour ago. On package it says SOLO-LV02. Had set it up when then found video on Youtube https://www.youtube.com/watch?v=UbbHBNUT1p4 how to flash the latest firmware and followed the guy. After flashing flashing firmware via recommended link it became dead. Then googled and found your threat. It 100% worked this guide and would like to thank you. You saved my shiny new miner!!

xIExodusIx, 2025-04-03:
I got a LILYGO T-DISPLAY S3 Nerdminer and tried to flash the prerelease 1.7.0. 
For the first try I took the three Lilygo-T-Embed files! --> Result: No funktion!
2nd try with the 3 NerdminerV2-T-Display_V1 files! --> Result: No funktion!
3rd try with the 3 NerdminerV2 files! --> Result: success! it works!