BitMaker-hub/NerdMiner_v2 #518: How to reset Wi-Fi settings

dneese, 2024-10-31:
How to reset Wi-Fi settings and Bitcoin wallet without reflashing the ESP32 WROOM board?

dneese, 2024-11-06:
I need to reset the settings, but without reprogramming the board. I'm not sure how to do this. Can someone help? Thank you!

darkfps, 2025-01-06:
> How to reset Wi-Fi settings and Bitcoin wallet without reflashing the ESP32 WROOM board?

Press the boot button for a few seconds.

The wroom wifi network should appear, so you can configure it again.

If not, you should repeat the flashing procedure described on the nerdminer home page

dneese, 2025-01-12:
> > How to reset Wi-Fi settings and Bitcoin wallet without reflashing the ESP32 WROOM board?
> 
> Press the boot button for a few seconds.
> 

Thank you, it worked!