BitMaker-hub/NerdMiner_v2 #535: heltec WiFi LoRa 32(V3) support

sdoros, 2024-12-06:
Is heltec WiFi LoRa 32(V3) supported? Can I try any of the f/w in https://flasher.bitronics.store?

Thanks!

JohnTaipei, 2024-12-19:
https://flash.nmminer.com/

sdoros, 2024-12-19:
Thank you for your answer, I am kindly asking if heltec V3 can be supported for Nerdminer v2, not for other miners that asking license.

JohnTaipei, 2024-12-20:
choose s3 dongle or t-qt
but no screen
oled no work

JohnTaipei, 2024-12-20:
you could play meshtastic
https://flasher.meshtastic.org/

sdoros, 2024-12-20:
> choose s3 dongle or t-qt but no screen oled no work

Thanks

closed 2025-01-13 by BitMaker-hub

codingneko, 2025-06-12:
@BitMaker-hub Hello, this is not completed, I suggest reopening this issue and planning a build for Heltec V3. This is a quite common board, and the fact that it has an integrated OLED screen is rather useful. There are also Heltec V3 versions that don't have LoRa capabilities, making them pretty inexpensive.