BitMaker-hub/NerdMiner_v2 #553: Issues facing with Lilygo ESP32-S3

GFreedomX, 2025-01-01:
hi guys 
I have some questions about the following device:
1)Lilygo ESP32-S3 T-Display 
2)Lilygo ESP32-S3 Dongle

For the Lilygo ESP32-S3 T-Display, I see it will drop to 0 Hashrate for some time and it will come back up. (Could this be it's getting a job or something?) and it tends to restart sometimes. 

For the Lilygo ESP32-S3 Dongle have the following issue
1)restart on its own randomly.
2)it will get stuck in asking to config using wifi like the initial setup after it restarts
3) if it loads up successfully, it will show zero hash rate. (The time is correct as to my initial config)
4) sometimes the display scrolls through very slowly as in if it lags

Lilygo ESP32-S3 Dongle are brand new which I just got hold of these for 2 to 3 days and is running off 1.7.0 version and even one 1.6.3 is also has the same issue

ESP32-S3 T-Display is about 1 year old and is running on 1.6.3.

GFreedomX, 2025-01-01:
i mange to get a log entry for it
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>false / subscribed>false / wificonnected>true
>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
>>> [i] Miner: newJob>false / inRun>false) - Client: connected>false / subscribed>false / wificonnected>true
>>> Completed 0 

This should be related to the zero hash rate for "3) if it loads up successfully, it will show zero hash rate. (The time is correct as to my initial config)" for the Lilygo ESP32-S3 Dongle. 

is there a coding written to resolve this?

Strykar, 2025-04-08:
I have the same issue on the original T-Display S3.