BitMaker-hub/NerdMiner_v2 #554: Worker nao appears unmineable.

hugohfsouza, 2025-01-06:
Personally, I noticed a few days that my ESP32 WROOM was working normally with the latest release. However, on (01/05/2025) the worker stopped appearing on the Unmineable site.

Our logs show that you are mining normally, but you are not crediting any more money.

I've been mining for more than 30 minutes and nothing appears

Does anyone know what could be happening?

hugohfsouza, 2025-01-06:
#update 

[image]

darkfps, 2025-01-06:
Notei isso tbm, me responderam o mesmo.
Descobri que a dificuldade tinha aumentado, cuidando via serial log do arduino IDE.

Ainda sem definição de nova moeda

gungriffonbr, 2025-01-12:
they fixing a new server for lower difficulty using 5555 port

CreddeDK, 2025-05-06:
Have anyone got this to work now? I cannot get mine to show up in the unmineable dashboard - trying different ports etc.

on 23/1 unmineable had the news:
"We’ve recently pushed an update to restore support for low hash rate devices (all ports) and improve pool compatibility. If you’re still experiencing connection issues, be sure to contact us via our support email. We’ll do our best to ensure your device is supported. Happy mining!"