BitMaker-hub/NerdMiner_v2 #556: ESP32 C3 Super Mini with screen

R9anSmart, 2025-01-07:
Hi,
Configured a C3 Super Mini successfully and managed to access in AP mode to configure.
Just wondering if there is a way that I can get the tiny oled screen functioning and/or is there a way confirm it is working and view the stats etc via web portal as currently I am only able to see it's working via network IP scan.

Is there also a way to chase the hostname of the device or edit the settings with it connected to a AP?

Thanks in advance

paelzer, 2025-01-07:
This one?

[image]

R9anSmart, 2025-01-07:
Thanks for fast reply. That's the puppy yes. Didn't want to give incorrect information by mentioning a model as very new to the world of esp devices

image.png (view on web)<https://github.com/user-attachments/assets/029da8da-3464-4d18-928f-e0e55631a8a7>

R9anSmart, 2025-01-07:
Not sure as I dont speak coding however looks like there are some useful details on what to add to the code to access the display etc in the reviews on the product page. I will also leave them below in case they are useful to you.

> The OLED is indeed a bit weird, you need a bit of trick to get it working: #include <U8g2lib.h> #include <Wire.h> // there is no 72x40 constructor in u8g2 hence the 72x40 screen is mapped in the middle of the 132x64 pixel buffer of the SSD1306 controller U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, 6, 5); int width = 72; int height = 40; int xOffset = 30; // = (132-w)/2 int yOffset = 12; // = (64-h)/2 void setup(void) { delay(1000); u8g2.begin(); u8g2.setContrast(255); // set contrast to maximum u8g2.setBusClock(400000); //400kHz I2C u8g2.setFont(u8g2_font_ncenB10_tr); } void loop(void) { u8g2.clearBuffer(); // clear the internal memory u8g2.drawFrame(xOffset+0, yOffset+0, width, height); //draw a frame around the border u8g2.setCursor(xOffset+15, yOffset+25); u8g2.printf("%dx%d", width, height); u8g2.sendBuffer(); // transfer internal memory to the display

There is also another:

> Install u8g2lib.h and wire H library and all the information entered from another buyer in the review I think from Netherland. Everything is OK. Write this code in Arduino IDE and it will work for you 100%. Little Wifi coverage but enough. The display shines quite poorly compared to OLED 128x64 but it is enough. I definitely RECOMMEND!!! Remember the offset of the screen (add X = 30 and y = 12) so that there is an image, text in OLED 70x40!

Hope this helps

R9anSmart, 2025-01-08:
At the moment, as it is headless, there is no real indication that it works at all. Could I also suggest that the blue light on the C3 models do a slow breath when working correctly, pulse when in AP mode and rapid flash for a win or something? Just a idea, not skilled enough to cook this kinda thing myself - grateful for all the work already put in by the team :)