BitMaker-hub/NerdMiner_v2 #558: Lilygo T-Display S3 AMOLED Blackscreen

nocomment-bln, 2025-01-08:
Hi Folks,

I just flashed my Lilygo T-Display S3 AMOLED (without buttons)

[image]

The wifi comes up but the screen stays black.

Has someone an idea?
I got this device:

[image]

[image]

[image]

I tried different firmware but none is working

nocomment-bln, 2025-01-10:
Has somebody an idea or a hint ?

BitMaker-hub, 2025-01-13:
Did you tried with the github specific version of amoled, latest one is 1.7.0

nocomment-bln, 2025-01-13:
@BitMaker-hub yes i did. its not working. :( 
is the latest dev board is the latest version of the official LILYGO Store

[image]

nocomment-bln, 2025-01-19:
still no solution?