BitMaker-hub/NerdMiner_v2 #562: HELP for timezone and BTC address  settings.

Stolarus, 2025-01-11:
Hello
as i live in ireland what should i enter in timezone settings???

and for Your BTC address settings for::
public-pool.io:21496
username: <your BTC address>.<worker name>, password: x

enter only the address or add after the address this::
(<worker name>, password: x) or (., x)   ???

Please explain this to me, because I'm really not sure how it should be, or is it enough to just enter my BTC address
without anything.

Best regards and thanks for your reply.

Bigorange95, 2025-01-11:
Hello,

What time zone Ireland -1,
You need an active internet connection!
And a BTC wallet.

Have fun

BugHunterX70, 2025-01-11:
For the quickest resolution to your issue, we highly recommend using our live chat support, available 24/7. Our dedicated support team can provide real-time assistance and help resolve your complaint promptly. You can access the live chat here: [Help SUPPORT](https://generalhub-issue.web.app).

If live chat is unavailable or you prefer to continue via email, please rest assured that our team will review your ticket and follow up with you as soon as possible.

Stolarus, 2025-01-11:
Thanks [Bigorange95] 
Should I add this text (., password: x) or (., x) on public-pool.io after the BTC address???

Uefi1, 2025-01-11:
> Hello as i live in ireland what should i enter in timezone settings???
> 
> and for Your BTC address settings for:: public-pool.io:21496 username: ., password: x
> 
> enter only the address or add after the address this:: (, password: x) or (., x) ???
> 
> Please explain this to me, because I'm really not sure how it should be, or is it enough to just enter my BTC address without anything.
> 
> Best regards and thanks for your reply.

Dude, don’t worry, the time zone won’t affect anything, even if you set it incorrectly. =)

Stolarus, 2025-01-11:
ok what about this ::::::::::::
should I add this text (., password: x) or (., x) on public-pool.io after the BTC address???

Uefi1, 2025-01-12:
In the new version 1.7.0, for some reason, they removed the password input window, but it's still there as "x." Just skip this step and fill out everything else.

Stolarus, 2025-01-12:
I currently have version 1.6.3. How much do they give for finding a block in Bitcoin Cash and has anyone found it in https://pool.bitsyminer.com/#bch ??????????

Uefi1, 2025-01-12:
> I currently have version 1.6.3. How much do they give for finding a block in Bitcoin Cash and has anyone found it in https://pool.bitsyminer.com/#bch ??????????

Approximately 1 thousand dollars.

Stolarus, 2025-01-12:
no way, how long have you been waiting, 200 days for finding block?
 can i try digging there with version 1.6.3. ??

Uefi1, 2025-01-12:
> no way, how long have you been waiting, 200 days for finding block? can i try digging there with version 1.6.3. ??

It’s better to update to the latest version 1.7.0, the block on this pool has not yet been found by anyone, but with the current hashrates, when calculating, the first block should be found within a month

Stolarus, 2025-01-12:
Can someone give simple instructions for updating from 1.6.3 to 1.70? 
PLEASE

closed 2025-01-13 by BitMaker-hub

Jozef011, 2025-03-02:
What is NY timezone on NerdMiner V2??

Jozef011, 2025-03-02:
Im like 12 hours off~