BitMaker-hub/NerdMiner_v2 #567: Is there any Support for run the NerdMiner onto ESP8266?

MajorDomDePIN, 2025-01-17:
Hello dear NerdMiner :)

Thank you very much for your efforts in developing this cool project

I have been trying for a few hours to get NerdMiner or LeafMiner running on an ESP8266 and unfortunately I have failed most of the time so far. First of all, after the reset after flashing, the LED of the MCU stayed on all the time but even then I couldn't find a new WLAN AP. 

So the question arises whether there is any compatibility at all between the current firmware and the ESP8266?

Thanks in advance for any answer

R9anSmart, 2025-01-17:
> Hello dear NerdMiner :)
>
> Thank you very much for your efforts in developing this cool project
>
> I have been trying for a few hours to get NerdMiner or LeafMiner running on an ESP8266 and [...]

Been doing the same over here and would also like to know the answer.
I managed to get the Leafminer flashed to the ESP866, but no luck with Nerdminer yet.

For Leafminer you need to download the following:

[image]
Download the file called "firmware_esp8266.bin"

Then flash it with: NodeMCU PyFlasher
https://github.com/marcelstoer/nodemcu-pyflasher

Look forward to finding out how to load Nerdminer onto the ESP866 however.

R9anSmart, 2025-01-17:
To add to this. My ESP8266 came with a integrated OLED screen. Is there support/can support be added for this?

MajorDomDePIN, 2025-01-18:
Thank you for the response, i was able to get it running with Leafminer, at least for some time.. sadly after a few mining circles the miner starts to stop producing valid hashes and starts to reconnect to the wlan. 

i tried to figure out whats going on when i looked into the source code, sadly i couldnt find the reason for the recon-issue, but i was able to find some lines there which refer to a screen inside of miner.cpp but they are in another if-case, so it won't get run on esp8266. 

i also couldn't find a config file for an esp8266 screen like i found it for the other mcu's:

[image]

maybe when you experienced with c++ you could add the feature by adding a proper screen config file and include its into the codebase

bobbikurniawan, 2025-06-27:
> Been doing the same over here and would also like to know the answer. I managed to get the Leafminer flashed to the ESP866, but no luck with Nerdminer yet.
>
> For Leafminer you need to download the [...]

I can't follow your tutorial, after I configure it, I connect to WiFi, but it disconnects for a moment, then reconnects for a moment, can anyone help me solve this problem?