BitMaker-hub/NerdMiner_v2 #569: Flash custom firmware

Gurenduben, 2025-01-19:
I've edited the code a bit, how do i flash the custom firmware?

Uefi1, 2025-01-19:
> I've edited the code a bit, how do i flash the custom firmware?

Hi can you change the algorithm from sha256d to sha256 ?

Gurenduben, 2025-01-19:
> > I've edited the code a bit, how do i flash the custom firmware?
> 
> Hi, can you change the algorithm from sha256d to regular sha256?

I can try 

[image]

Uefi1, 2025-01-19:
> > > I've edited the code a bit, how do i flash the custom firmware?
> > 
> > 
> > Hi, can you change the algorithm from sha256d to regular sha256?
> 
> I can try
> 
> [image]

Please make it compatible with ESP32-D0WD-V3-weact, I would love to try it since most pools require regular sha256!

Gurenduben, 2025-01-19:
> > > > I've edited the code a bit, how do i flash the custom firmware?
> > > 
> > > 
> > > Hi, can you change the algorithm from sha256d to regular sha256?
> > 
> > 
> > I can try
> > [image]
> 
> Please make it compatible with ESP32-D0WD-V3-weact, I would love to try it since most pools require regular sha256!

Im not the developer of this software

Gurenduben, 2025-01-19:
> > > > > > I've edited the code a bit, how do i flash the custom firmware?
> > > > > 
> > > > > 
> > > > > Hi, can you change the algorithm from sha256d to regular sha256?
> > > > 
> > > > 
> > > > I can try
> > > > [image]
> > > 
> > > 
> > > Please make it compatible with ESP32-D0WD-V3-weact, I would love to try it since most pools require regular sha256!
> > 
> > 
> > Im not the developer of this software
> 
> I know, but if you understand the code, just compile with regular sha256

Ok I'll edit and let you know if it works

Uefi1, 2025-01-19:
> Ok I'll edit and let you know if it works

Yes, I will test it myself on some pools that use the regular sha256 algorithm and provide feedback on the results.

closed 2025-01-20 by Gurenduben