BitMaker-hub/NerdMiner_v2 #571: v1.70 prelease for nerdminer2_tdisplays3 not available on https://flash.nmminer.com

dwightmulcahy, 2025-01-23:
I don't see the image for Lilygo T-Display S3  under the dropdown for images on https://flash.nmminer.com/

Uefi1, 2025-01-24:
First of all, you are working on the wrong website to create the firmware. Secondly, T-Display v1 should work.

https://flasher.bitronics.store/

dwightmulcahy, 2025-01-24:
Actually was using the nmminer flasher to only connect for the console without flashing, something that https://flasher.bitronics.store/ lacks.

dwightmulcahy, 2025-01-24:
@Uefi1 your image shows V1.6.3... I am trying to flash the prerelease V1.7.0.  The naming isn't consistent.

I tried a some the S3 versions without success.

[image]

Uefi1, 2025-01-24:
> [@Uefi1](https://github.com/Uefi1) your image shows V1.6.3... I am trying to flash the prerelease V1.7.0. The naming isn't consistent.
> 
> I tried a some the S3 versions without success.
> 
> <img alt="Image" width="484" src="https://private-user-images.githubusercontent.com/8507802/406527561-97f4fff8-8563-4cdf-b95a-f0e9485b3982.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzc3NTg4NzUsIm5iZiI6MTczNzc1ODU3NSwicGF0aCI6Ii84NTA3ODAyLzQwNjUyNzU2MS05N2Y0ZmZmOC04NTYzLTRjZGYtYjk1YS1mMGU5NDg1YjM5ODIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMjRUMjI0MjU1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTExMWYyMmQzYjNiNjdiMmFkOTg3NDZkZjI4Y2NmMWIwYzhhYTE4ZjhiOGM2NGQ3NmJhNzYyZTcwOWY2ODA1NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.yt8IKYrPfb8cuBaeIFys2OHpAAespu6hVC3aA7XbBLY">

In NerdMiner, versions are calculated differently: 1.6.3 there is equivalent to 1.7.0. =)

schwatter, 2025-08-10:
The same for me. Last working version ist 1.6.3. I tried all after that. But no one is booting at my Liligo T-Display.