BitMaker-hub/NerdMiner_v2 #579: Suddenly stopped mining

davea38, 2025-02-02:
Hi,

I've been running two NerdMiner v2's for nearly two months. Woke up today to 0 hash rate. I checked with a friend of mine who also has a NerdMiner and she has 0 hash rate too.

All the miners mentioned above are setup using the default pool (which I believe is publicpool.io). I've tried pinging publicpool.io and it responds, plus I can tracert it fine.

Can anyone help??

jpduhen, 2025-02-02:
Same here, zero hash

davea38, 2025-02-02:
That's the one. I'm new to NerdMiner - do these pools go up and down much? Should I maybe move to another pool

TylonHH, 2025-02-03:
No that's very rare. But there are also many other pools.

Public Pool is up and running so this issue here should be solved?

davea38, 2025-02-03:
Yup all back up and running now, thanks for everyone's help :)

closed 2025-02-03 by davea38