BitMaker-hub/NerdMiner_v2 #583: Wifi Connection Issue

cher3017, 2025-02-06:
Hello

I have the ESP32 WROOM 32D with 1.6.3 fireware and doesn't connect to my Wifi. 

I update the Wifi and Pool configuration, but doesn't work.

Any idea?

Thanks for your support!!

julianobd, 2025-02-07:
I didnt understand very well your issue.

is your Wifi displayed on wifi manager?
you connect to wifi and dont start mining?

cher3017, 2025-02-07:
Hello

No, I don't have display, when I finish to configuration the NerdMimerAp Wifi and reboot the miner doesnt connect with my Wifi and doesnt start miner.

Thanks

julianobd, 2025-02-07:
As the device is restarting and the access point is being disabled, it probably connected to WiFi. Which pool are you trying to connect to? Some pools are not compatible with NerdMiner due to its low hashrate.

stefi01, 2025-02-15:
I had the same issue with the same board, my fix came from a video and he has a link to the file in his description, [https://www.youtube.com/watch?v=gGPa_K47aL0](url)