BitMaker-hub/NerdMiner_v2 #588: Very low range in 20 boards (probably board defect)

ChicoCybergoth, 2025-02-11:
Hello miner friends.

I have a problem here, I would like to see if someone can help me.
I bought 20 ESP32-WROOM modules, 10 from one seller and 10 from another on different dates, all with internal antenna.
ALL 20 modules have a terrible range.
Some of them do not reach 5 meters in the visual field.
At least half of them cannot reach 5 meters with a wall in between.

I took some measurements with an app, and with one of them I got this:
With the cell phone touching the board: -50dBm
At 5 meters away: -88dBm
At 5 meters away with a wall in between: No range.

It is probably a defect in the modules, but it is strange that 100% of the modules have the same defect even though they were bought from different sellers and on different dates.

These were the modules I bought:
aliexpress.com/item/1005006476877078.html
aliexpress.com/item/1005004605399313.html
I had already bought one of them in the past and didn't have the problem (or didn't notice the problem).

I have already prepared more than 50 of these modules and had never experienced this.

I tested it with a module from an older purchase, perhaps purchased from the link suggested here on Nerdminer's Github (which currently no longer provides the WROOM model).

I measured -80dBm at about 9m with two walls in between, it performs MUCH better than the modules I'm having problems with.

Does anyone have any idea what this could be?
Thank you very much.

Uefi1, 2025-02-12:
No, this is not a defect at all. Initially, the ESP32 with an internal antenna indeed has very poor Wi-Fi signal reception. I soldered an external antenna connector following this video.

https://youtu.be/o4mtrueU6eM

ChicoCybergoth, 2025-02-12:
@Uefi1 do you really think that a module like this with a range of less than 5 meters in the visual field is normal?

Honestly, I don't really believe it, especially because I have other good modules here that have a normal range. I put them side by side and did a lot of tests. While the good module is at -34 dBm, the bad modules are at -70 dBm and even -80 dBm, all side by side, 30 cm from the cell phone that is taking the measurement.

I believe that a range of less than 5 meters in the visual field is completely unrealistic.
Besides, having to repair something 100% new doesn't seem right.

The curious thing is that, powering the module through the pins instead of through the USB gives a great increase in the range.