BitMaker-hub/NerdMiner_v2 #591: Feature Request - Automatic switch between summer and winter time in europe

DragonPerl, 2025-02-18:
Hi Markus here!
I am enjoying my NerdMinerV2 Clock on my Office Desk.
So thanks for all the effort you put into this project.

I'm not that good at programming so I am sorry to say that I can't make the changes myself.

So it would be great to have the option of automatic switching between summer and winter time so i don't have to reset my nerdminer twice a year.

It doesn't effect mining thou but it is irritating to have the wrong time displayed on my desk.

Thanks a lot!
BR Markus

m8ax, 2025-03-07:
Prueba este firmware https://m8ax.github.io/MvIiIaX-NerdMiner_V2-DeV/

Lab1717, 2025-03-08:
> Probe this firmware for nerdminerv2 https://m8ax.github.io/MvIiIaX-NerdMiner_V2-DeV/

Too bad your repo does not speack English.

m8ax, 2025-03-08:
why my reoo must be in engish ?

Lab1717, 2025-03-08:
> why my reoo must be in engish ?

Hi, i did not said it must be in english, I said too bad it is not in English.