BitMaker-hub/NerdMiner_v2 #592: Feature Request - Support for Waveshare esp32s3 lcd 1.47 inch dongle

DragonPerl, 2025-02-18:
I have this board lying around and it would be amazing if I could turn this into a NerdMiner.

I saw it on another site where they sell this board with the firmware already ported, but i can't find the code anywhere.
So it would be nice if the guys selling it would provide the code to this open source project, as they profit from it.

https://bitcoinmerch.com/products/nerdminer2?variant=45551650537702

Thanks a lot!
BR Markus

DarKOrange75, 2025-03-20:
actually it is working with **nerdminer-prerelease-V1.7.0_NerdminerV2-S3-AMOLED** but the screen stays black... anyone could help please?

theklunky, 2025-04-22:
I appreciate the response to this issue but it's way above my abilities to make that work lol. Any chance you can provide more details? I am useless at C dev. If it was in python, I could probably figure it out hahaha

theklunky, 2025-04-22:
Wow, you delivered!! Unfortunately it fails to build. Lots of errors. The compile output was too large for here so I uploaded it here: https://pastebin.com/ZjUg1FVC 
Any tips on next steps?

BlueVenom1, 2025-04-23:
Same... Used VS to try and build. So many errors. Anyway to just upload the finished files if anyone gets it to work?

Cjohnson233, 2025-05-22:
Cannot read properties of undefined (reading 'id')

has anyone figured this out??

odroegehorn, 2025-08-13:
Hey all,
great project. I also have an ESP32-C6  1.47" display device lying around.
Can somebody compile a working binary to flash?
I have the LilyGos working, but the ESP32-C6 is so nice and slim. 
Thanks.