BitMaker-hub/NerdMiner_v2 #630: Compatibility Inquiry: Does NerdMiner Support parasite.wtf? Hash Data from parasite.wtf Not Syncing

ifree321, 2025-05-14:
I connected my nerd miner to parasite.wtf and obtained hash data, but I cannot find my data on parasite.space. Can you see it? I even changed the address to bc1qzkw9wgdgks468p8g9m0yk3k4ppnsn74dqy9ps3.bboy.3effa696bf@sati.pro@parasite.sati.pro or bc1qzkw9wgdgks468p8g9m0yk3k4ppnsn74dqy9ps3.bboy.3effa696bf@parasite.sati.pro, but I still can't find it.

Miner Setup Guide From https://parasite.space

Set up a new private key on Xverse wallet:
Create a brand new wallet (not just a new account)
Ensure there are no Bitcoin or Ordinals on it
Securely store your private key
Visit [parasite.sati.pro](https://parasite.sati.pro/) and connect your new Xverse wallet:
Complete the wallet connect signature
Copy your generated "static ln address" for later use
Connect your Bitcoin miner and go to the "Pool Settings" tab
Configure your mining settings:
Stratum URL: parasite.wtf
Port: 42069
Set your username using this format:
xverseL1address.username.staticlnaddress@parasite.sati.pro
Example: bc1qnotarealaddress.steve.yourreallnurlhere@parasite.sati.pro

Use any password (it's not checked)
Save your configuration and restart if needed
Participation in Bitcoin mining, including through Parasite Pool, which is still considered in beta testing, involves risks such as market volatility, hardware failure, and changes in network difficulty. Parasite Pool is in beta and has not yet found a block; there is no assurance of future block discoveries or payouts. Users should exercise caution and consider their financial situation before engaging in mining activities.

ifree321, 2025-05-15:
I just found out that the execution logic of nerdminer truncates the address to 80 characters, while parasite requires at least 82 characters. I changed it to 100, but I still can't find it on the website. Perhaps it's because the 32-bit share hasn't been submitted yet. I'll wait a bit longer.

ifree321, 2025-05-20:
Even if the configuration is correct, it is still impossible to view worker information from parasite.space. Is there any solution? Does anyone understand?

BitMaker-hub, 2025-09-16:
Hi, I'll test it whenever I can. But its possible that current pool dificulty would be higuer than what Nerdminer can deliver on a window of time were the socket with pool is still open.
I suggest using one of the low diff pools shown on repo description