BitMaker-hub/NerdMiner_v2 #671: Significant Hashrate Drop After Updating to nerdminer-release-V1.8.1 (from 300KH/s to 55KH/s) on ESP32-2432S028

leochan2017, 2025-09-08:
I have a yellow ESP32-2432S028 board with a screen. When I flashed the nerdminer-prerelease-V1.8.1 Pre-release firmware, it achieved a hashrate of 300KH/s. However, after updating to the latest nerdminer-release-V1.8.1 version today, the hashrate dropped dramatically to only 55KH/s. What could be causing this performance degradation?

andrewjswan, 2025-09-08:
Look like this https://github.com/BitMaker-hub/NerdMiner_v2/pull/667/

leochan2017, 2025-09-09:
> Look like this [#667](https://github.com/BitMaker-hub/NerdMiner_v2/pull/667)

Okay, maybe 55KH/S is the real hash value of this board at the moment. 😂

But strangely, my other boards like ESP32-C3 Super Mini can still maintain 300KH/S. 😅

BitMaker-hub, 2025-09-09:
Yes, this is related to #667 
unfortunatelly as I could see there was a misscalcultion on standard ESP32 devices. On current ESP32-C3, S3, or S2 should not be any problem. Current HW acceleration functions work good

andrewjswan, 2025-09-09:
> unfortunatelly as I could see there was a misscalcultion on standard ESP32 devices.

It's very strange, on the old version, if you believe the pool, then ESP32 shows about 300 with a fairly large uptime.

andrewjswan, 2025-09-09:
[image]

BitMaker-hub, 2025-09-09:
Ok, can you tell me the board you are using and got this hashrate. I'll retest again because I saw real differences.

BitMaker-hub, 2025-09-09:
Guys, just released 1.8.2 with reverts and new fixes, you can flash it at [flasher.bitronics.store](url)

tx for your feedback

closed 2025-09-09 by BitMaker-hub