BitMaker-hub/NerdMiner_v2 #687: Dark mode?

techrivertree, 2025-09-14:
My first ESP miner is from BTC merch and Nerd Miner V2 is in dark mode. My new ESP miner is stuck on lite mode which hurts my eyes. I dug through the settings but I could not find dark mode.

So, how can I enable dark mode on my ESP32?

techrivertree, 2025-09-16:
If you look at the lottery miners from BTCmerch https://bitcoinmerch.com/, they are running nerd miner 2 in dark mode and over at flasher.bitronics, the nerd miner v2s in background are in dark mode.

[image]

ArminXXXXX, 2025-09-27:
i have the same issue. the display is not black it is white in the too. how can I change this?